sinara-hw/Pounder #103: DDS baluns/termination are wrong

jordens, 2022-07-13:
The source output impedance after the baluns (TR9 etc) is 100 Ohm currently. The filters want 50 Ohm. (The eval board also has a 100 Ohm single ended output impedance from DAC1_OUT into 50 Ohm filters/SMA output which doesn't sound optimal).
We can either make the source termination R24...R35 25R or use TC2-1TX+ as baluns. The latter gives more power and is still within DAC compliance.

gkasprow, 2022-07-28:
fixed
[image]

closed 2022-07-28 by gkasprow

jordens, 2022-08-01:
Just to remind myself of the context:

Datasheet says:

[image]

While the text is consistent, the figure is wrong.
Note the missing resistors at the source side of the balun. The Source would be ~100 k impedance.

Eval board has:

[image]

This is also wrong.
The 50 R are populated. Hence the source impedance is 100 R. The balun is 1:1. DAC_OUT is 50 R (filter and SMA)

Various ANs from ADI with different information:

https://www.analog.com/media/en/technical-documentation/application-notes/an_912.pdf looks trustworthy and is consistent.

[image]

https://www.analog.com/media/en/training-seminars/tutorials/MT-019.pdf similarly consistent.

[image]

https://www.analog.com/media/en/training-seminars/design-handbooks/Technical-Tutorial-DDS/Section6.pdf is wrong.

[image]

The AD9850 is also an unterminated current source with 100 k impedance.

Some Analog forum posts indicating the obvious, that this is indeed a current source and the source termination should match the load impedance:

https://ez.analog.com/dds/f/q-a/536170/ad9959-output-impedance
https://ez.analog.com/dds/f/q-a/28841/ad9959---no-ac-output-improper-termination